mph-/lcapy #93: Drawing a choked inductor

odkken, 2022-11-01:
Hi, I'm trying to insert a choked inductor into a circuit for the sole purpose of drawing:
[image]

I'm able to render that using
```
cct = Circuit("""
;;\draw (0,0) to[cute choke, twolineschoke] ++(2,0);
""")
cct.draw()
```

but am hoping there's a way to get this symbol using normal lcapy circuit symbols:

```
cct = Circuit("""
A1 0; l=\mathrm{VSW\_ABC}, anchor=east
L6 0 1 .00000033; #<-- make this choked
;label_nodes=false""")
cct.draw()
```
[image]

Is there a semi-painless way to substitute out symbols like this, or change the style on individual components?  I tried playing with the example of ``` ;; \ctikzset{inductor=cute}```
which works for switching between the american/cute styles, but couldn't change to choked versions.

mph-, 2022-11-01:
I have just pushed a patch to support this.  This now supports the following:

    L1 1 2; right, kind=choke
    L2 2 3; right, kind=twolineschoke

odkken, 2022-11-02:
Amazin!  Thanks.

closed 2022-11-02 by odkken