mph-/lcapy #94: Transfer function calculated wrong?

oe8bck, 2022-11-03:
I think the K-factor in this small circuitry is wrong, The K-factor given is K=-R1/(C2R2^2)
I do not expect a capacitor (Farad) in there, all units should cancel.

```
from lcapy import Circuit, s, oo

schem=Circuit("""
sch1.W 8_1 8_2; up
sch1.R1 8_2 10_1; right,l=R_1
sch1.W 10_1 10_2; right
W sch1.outp outp; right=1
sch1.W outp 10_2; up
E1 sch1.outp 0_5 sch1.8_1 0 -A0; down, l=-A_0 \cdot vgnd_{n}
A11 sch1.8_1;l=vgnd_n,anchor=west
W 0_5 0; down=0.1,implicit, anchor=east, l=V_{SS}
R_21 sch1.8_1 14; down, l^=R_2
C2 14 0_6; left, l^=C_2
W15 0_6 0; left=0.1,implicit, anchor=east, l=V_{SS}
R_22 14 14_1; down, l^=R_2
W14 14_1 vin_n; left
A15 vin_n;l=vin_n,anchor=east
A16 outp;l=vout_p,anchor=west
;help_lines=0,label_nodes=none,scale=0.25
""")

schem.draw()

H = schem.transfer('vin_n', 0, 'sch1.outp', 0)
Hzpk=H.subs('R_22','R2').subs('R_21','R2').limit('A0', oo).ZPK()

display(Hzpk)

X1, defs = Hzpk.parameterize_ZPK()
display(defs)
Hzpk.subs({'R1':10e3,'R2':100e3,'C2':10e-9}).bode_plot((1, 10e6),title="Hall path")
```

Many thanks for the hint what's going wrong.

KR,
Christof

oe8bck, 2022-11-04:
Thanks for your fast response. 

Well, to be honest up to now I always thought that K is always dimensionless. But you are absolutely right, if the number of poles is not the same as the number of zeros, the units of poles and zeros do not cancel and so - if the transfer function is dimensionless (Vo/Vi) - the K-factor has to have a unit.

Thanks a lot for that great package, I just started to work with it and I like it more the more I am using it :-)

mph-, 2022-11-04:
I find circuit behaviour is full of surprises!  

I've fixed expression parameterization so that the units are correctly determined for the parameters.  So for your example:

    >>> defs['K'].units
    rad
    ──
     s