mph-/lcapy #95: Broken symbolic result displaying in Jupyter Lab

denis-x, 2022-11-11:
Hello,

It looks like, there some sympy hook, used in LCAPY, breaks symbolic result displaying in Jupyter Lab (Windows).

E.g. cell with content
> from lcapy import t
> t
get stuck in execution state [*] for infinite time.

But
> from lcapy import t
> str(t)
works well.

Is any idea, what it can be related?

BR Denis

mph-, 2022-11-11:
Hmmm, this works for me with Jupyter on Linux.  Are you able to check if it works for you with Jupyter?  Could you also please post the output from:

    from lcapy import show_version
    show_version()

denis-x, 2022-11-14:
Hello,

Python: 3.8.13 | packaged by conda-forge | (default, Mar 25 2022, 05:59:45) [MSC v.1929 64 bit (AMD64)]
SymPy: 1.10.1
NumPy: 1.22.3
Matplotlib: 3.5.1
SciPy: 1.8.0
Lcapy: 1.8

denis-x, 2022-11-14:
Even if upgrade to SymPy: 1.11.1, I do not see any improvement

# Below works well
import sympy as sp
sp.S('t')

# Below gets stuck at execution
import sympy as sp
import lcapy
sp.S('t')

mph-, 2022-11-14:
Has this ever worked for you or is it a new problem?

It works for me with either Sympy 1.10.1 or SymPy 1.11.1 using Jupyter or JupyterLab on Linux.

Did you try your example using Jupyter; not JupyterLab? 

FYI, Lcapy overrides SymPy's printing methods in lcapy.printing.py

denis-x, 2022-11-14:
Yes, the same.

Python: 3.9.7 (default, Sep 16 2021, 16:59:28) [MSC v.1916 64 bit (AMD64)]
SymPy: 1.9
NumPy: 1.20.3
Matplotlib: 3.4.3
SciPy: 1.7.1
Lcapy: 1.2.3

denis-x, 2022-11-14:
Still in place after upgrade
Python: 3.9.7 (default, Sep 16 2021, 16:59:28) [MSC v.1916 64 bit (AMD64)]
SymPy: 1.11.1
NumPy: 1.20.3
Matplotlib: 3.4.3
SciPy: 1.7.1
Lcapy: 1.9.1

mph-, 2022-11-14:
But has it ever worked for you or is it a new issue?

mph-, 2022-11-14:
Ah, perhaps it is a Windows problem for you.

Do you have mathjax installed?

denis-x, 2022-11-14:
Yes. And
import sympy as sp
sp.S('t')
displays me math, as well as, other ways of using sympy and pandas for outputting expression.

And I can get version number by right click on displayed formula
MathJax.js v2.7.9
MathML Input Jax v2.7.9
TeX Input Jax v2.7.9
HTML-CSS Output Jax v2.7.9
NativeMML Output Jax v2.7.9
PreviewHTML Output Jax v2.7.9
mml Element Jax v2.7.9

mph-, 2022-11-14:
I'm home crook with a cold so will get one of my students to try on Windows tomorrow.  It must be something subtle.

mph-, 2022-11-15:
Yes, it is Windows specific.  I have had to open up the SymPy printing can of worms and found that the some of the printing classes have changed.

Could you try checking out the latest version of Lcapy from git, do a `pip install  . ` in the root dir of Lcapy, and see if the printing is fixed?  Thanks.

denis-x, 2022-11-15:
Hello,

I have got it: Lcapy: 1.10.dev0, but the issue has not gone.

mph-, 2022-11-15:
The plot thickens.  I tried using Jupyter on Windows with Firefox and it worked!

Which browser are you using?

denis-x, 2022-11-16:
Hello,

Unfortunately, on my side, it is browser agnostic. I have run Juputer Notebook with different browsers and got the same behavior with Firefox, Chrome, MS Edge.

mph-, 2022-11-17:
Thanks for your patience and for trying this out.  Unfortunately it does not fail for me on Linux or Windows so I can't dive in with the debugger to find out where things are going wrong.  Both the Jupyter and SymPy printing machinery are rather complicated with fallback strategies depending on what programs are installed.

Anyway, I have found a problem with a function signature due to a change in the SymPy printing API.  I have pushed a patch for this.  Hopefully this will fix the problem :-)

mph-, 2022-11-22:
I haven't been able to recreate this problem on other computers running Windows.

All I can suggest is running the command `t.pdb()`.  This will throw you into the Python debugger.  Then insert a breakpoint `b latex.__call__`, continue, and then step to see what is hanging...

denis-x, 2022-11-23:
Thanks for suggestion, I was able to trace it to _get_latex_main, then I cannot proceed and step in into this function as debugger is stuck

ipdb>  s
> c:\users\nxa14730\appdata\roaming\jupyterlab-desktop\jlab_server\lib\site-packages\sympy\printing\preview.py(294)preview()
    292         raise OSError("Unrecognized viewer: %s" % viewer)
    293 
--> 294     latex_main = _get_latex_main(expr, preamble=preamble, packages=packages,
    295                                  euler=euler, extra_preamble=extra_preamble,
    296                                  fontsize=fontsize, **latex_settings)

ipdb>  s
> c:\users\nxa14730\appdata\roaming\jupyterlab-desktop\jlab_server\lib\site-packages\sympy\printing\preview.py(296)preview()
    294     latex_main = _get_latex_main(expr, preamble=preamble, packages=packages,
    295                                  euler=euler, extra_preamble=extra_preamble,
--> 296                                  fontsize=fontsize, **latex_settings)
    297 
    298     debug("Latex code:")

ipdb>  s
> c:\users\nxa14730\appdata\roaming\jupyterlab-desktop\jlab_server\lib\site-packages\sympy\printing\preview.py(294)preview()
    292         raise OSError("Unrecognized viewer: %s" % viewer)
    293 
--> 294     latex_main = _get_latex_main(expr, preamble=preamble, packages=packages,
    295                                  euler=euler, extra_preamble=extra_preamble,
    296                                  fontsize=fontsize, **latex_settings)

ipdb>  s
Output of this cell has been trimmed on the initial display.
Displaying the first 50 top outputs.
Click on this message to get the complete output.

denis-x, 2022-11-23:
ipdb>  a
expr = '$\\displaystyle t$'
output = 'png'
viewer = 'BytesIO'
euler = False
packages = ()
filename = None
outputbuffer = <_io.BytesIO object at 0x000002DE319735E0>
preamble = None
dvioptions = ['-T', 'tight', '-D', '150', '-bg', 'Transparent', '-fg', 'Gray', '-O', '0cm,0cm']
outputTexFile = None
extra_preamble = ''
fontsize = '10pt'
latex_settings = {}

denis-x, 2022-11-23:
On Linux (below) I have the same inputs, except for expr, could '$\' instead of '$\\' make a problem?
expr = '$\\displaystyle t$' output = 'png' viewer = 'BytesIO' euler = False packages = () filename = None outputbuffer = <_io.BytesIO object at 0x7ff1103971d0> preamble = None dvioptions = ['-T', 'tight', '-D', '150', '-bg', 'Transparent', '-fg', 'Gray', '-O', '0cm,0cm'] outputTexFile = None extra_preamble = '' fontsize = '10pt' latex_settings = {}

denis-x, 2022-11-23:
Sorry, maybe wrong point, github converts double slash into single and single just disappears.

mph-, 2022-11-23:
Thanks for this.

It seems like the problem is with the SymPy preview and that you don't have a program installed so it tries a fallback approach.

Can you try printing 0 * t + 3 and see if that works?

Have you installed the latest code from github I pushed last week?

denis-x, 2022-11-24:
Hmm, no printing is working for any output, neither '0 * t + 3' nor basic 'a=0, a' or simply '0'.
I have checked a bit more and found that issue happens during subprocess call of following command
'C:\\texlive\\2022\\bin\\win32\\dvipng.EXE', '-T', 'tight', '-z', '9', '--truecolor', '-o', 'texput.png', 'texput.dvi'
of preview/_check_output_no_window. 

I have tried to execute it manually and seems there is some problem with input .dvi
PS C:\Users\nxa14730\AppData\Local\Temp\tmp8lhgqq_r> C:\\texlive\\2022\\bin\\win32\\dvipng.EXE -T tight -z 9 --truecolor -o texput.png texput.dvi                                                         
This is C:\texlive\2022\bin\win32\dvipng.EXE 1.17 Copyright 2002-2015, 2019 Jan-Ake Larsson
[1 <raw PostScript

denis-x, 2022-11-24:
I have it further debugged and found, that on my side dvipng is not truck if pass it "--nogs" option.
When I have forced this option in preview/_check_output_no_window, I have got an output.
It looks like LCAPY has own options to call dvipng, could you, please, add "--nogs" to them, if there is not other objectives.

denis-x, 2022-11-24:
Yet another point from my side. Normally in jupyter, if type e.g. '0' in cell, it produces textual '0' in cell's output.
But if I import LCAPY, such simple '0' is run through sympy/preview, latex + dvipng, and picture is displayed in cell's output.

mph-, 2022-11-25:
That's great progress but it still has me puzzled.

What really puzzles me is that you said that SymPy printing works for you.  However, Lcapy's printing is a thin wrapper on SymPy printing and neither Lcapy nor SymPy uses the `--nogs` option for dvipng.   

I can explain why '0' is printed using LaTeX when Lcapy is imported.  This is because Lcapy calls the SymPy `init_printing` function.  This hooks into the Jupyter printing so that the latex printer is used to display the cell output.

denis-x, 2022-11-25:
SymPy utilizes some different routing (not preview) to display formulas. It explains, that I have not experience any problem with it.

When I have called SymPy preview with PNG & BytesIO, Jyputer gets stuck at dvipng call.
LCAPY utilizes preview and its dvipng call for everything, so Jupyter get stuck for any output, even '0' or some variable.

mph-, 2022-11-25:
I think I have it figured it out.  

When printing, Jupyter generates a number of different output representations, plain, png, latex, svg, etc.  However, SymPy disables the png and svg representations for its objects.  Lcapy did not disable the png representation so it triggers a bug in SymPy when converting to png.   

I've pushed a fix for Lcapy expressions to disable png ouput but will need to do this for lcapy container types.

closed 2023-01-19 by mph-