mph-/lcapy #98: Example in "Circuit analysis for novices" encountered error

vsardebili, 2023-02-26:
Following isues were encountered on page: https://lcapy.readthedocs.io/en/latest/novice.html
Running  on * Spyder version: 5.4.0  (pip)
* Python version: 3.11.1 64-bit
* Qt version: 5.15.2
* PyQt5 version: 5.15.7
* Operating System: Windows 10

1- state.abbreviate_units=False did not result in full unit display. Probably any issue within lcapy package.
2- a.R.v.subs(omega0, 3).plot((-1, 10)) resulted in "NameError: name 'omega0' is not defined". 
However, using a.R.v.subs('omega_0', 3).plot((-1, 10)) was successful in creating the plot. The page may need updates to match the latest version of lcapy.

Cheers
Vahid

vsardebili, 2023-02-26:
I am interested in contributing to this project. However, I am new to the community. When I followed the link at the top of the above link I was directed to the doc page in edit mode. I am wondering what are the rules for making the minor correction related to second sub-issue myself?

mph-, 2023-03-29:
Sorry for the slow response; I've just found the notification in my email junk folder.

If you wish to help fix up the docs, you will need to clone or fork Lcapy.  The doc files are in the directory lcapy/docs.  These are written using restructured text and are built with sphinx.  If you use a shell, typing `make doc` will build them if you have sphinx installed.  The docs can be locally viewed by pointing your  web browser at `lcapy/doc/_build/html/index.html`.

If you have forked the project you can commit your changes and send me a pull request.  Otherwise, send me the output from `git diff`.

Regards, Michael