mph-/lcapy #99: Simple update in installation instructions would help novices join the community more readily

vsardebili, 2023-02-26:
Suggested enhancement in installation document to help novice potential users who are not familiar with LaTeX: https://lcapy.readthedocs.io/en/latest/install.html#:~:text=For%20schematic%20drawing,pdflatex

Editing:
1. pdflatex
to:
1. pdflatex: Please see https://www.latex-project.org/get/
Would help less experienced persons willing to get started. 

Moreover, it could be useful to update: https://lcapy.readthedocs.io/en/latest/install.html#:~:text=install%20collection%2Dlatexextra-,Installation%20for%20Windows,Download%20circuitikz%20(https%3A//www.ctan.org/pkg/circuitikz),-Installation%20of%20latest
accordingly to include pointers to major windows downloads.

Cheers
Vahid

vsardebili, 2023-02-26:
This may relate to issue #89.

mph-, 2024-02-12:
Sorry, I've overlooked this issue.  I've added your suggestion about pdflatex but I have absolutely no competence with Windows.  Could you recommend which download?