mph-/lcapy #104: Is it possible to support logic gate schematic?

hazzm02, 2023-04-09:
I have tried lcapy to draw some Integrated Circuit schematic.  It is elegant to just have some barebone lines while still able to get what i want. Thank you very much Hayes!

When it comes to digital integrated circuit designs, Lcapy seems not able to draw AND, NAND, NOR, XOR symbols.
Can lcapy have an enhancement to make use of the logic gates from circuitikz?

Example:
combinational Gates:
[image]

mph-, 2023-04-11:
Yes, I have never gotten around to add these components.  They should be straightforward to add.  

I think the syntax for the two input, one output gates would be something like:

    U1 or
    U2 and
    U3 nor
    U4 nand
    U5 xor
    U6 xnor
    U7 equiv

I guess for multiple inputs, the syntax could be extended with `U8 and8`, `U9 or4` etc.
What do you think?

mph-, 2023-04-11:
I think I would use the circuitikz macro to draw the gate, just like you did.  The only problem is that the circuitikz components  get tweaked in size occasionally.

The case of the name is not an issue; we could support both.

mph-, 2023-04-12:
Did this patch do what you want?  Note, scaling does not work yet.

hazzm02, 2023-04-13:
Hi Hayes, 

Thank you for the quick action!
Yes, it is what i expected. Thank you very much!

But it is strange that embeded latex draw line is missing. 
ff2 Q pin to g1 in2 pin
[image]
`
from lcapy import Circuit
cct = Circuit("""
U1 dff; right, l=FF1
U2 dff; right, l=FF2
U3 dff; right, l=FF3
U4 and; right, l=g1
W U1.q U4.in1; right
W U4.out U3.d; right
O U1.mid U2.mid; down=1.3
;; \draw (U2.q) -| (U4.in2) ;
""")
cct.draw(draw_nodes=False,label_nodes='alpha')
`