mph-/lcapy #108: Mechanical Circuit Simulation

hrtlacek, 2023-08-14:
Not 100% sure but this seems like a bug (or missing feature). If I attach a voltage source to what should be the equivalent of an RC lowpass I encounter weird behavior.
I am not expert in this but I am sending  a heaviside function through a spring and dashpot and i get a heaviside function.
Probably I should just wait for force and velocity sources and not just expect the voltage source to work as a force source. 

Reproduce:
```
b = Circuit("""
V1 1 0 {u(t)}; down
P2 3_0 10_0; down, v=f_o(t)
W 0 10_0;
W 1 1_0; right
#r2 1_0 2_0 2; right, offset=1.5
r1 1_0 2_0 2; right, offset=0.5
k1 1_0 2_0 0.25; right, offset=-0.5
W 2_0 3_0; right
W 0 0_15; down=0.2, ground
;color = gray
""")
b.annotate_node_voltages().draw()
```

mph-, 2023-08-14:
Yes, in hindsight this was a mistake.  I chose the mechanical analogue II (with force analogous to voltage) since I was modelling transducers at the time and this was the convention.  However, this is really confusing with a nodal based system since you need to draw the mechanical circuit with one topology and simulate it with another!

It probably makes more sense to switch to the other analogue (with force analogous to current).   This preserves the circuit topology.

I welcome any suggestions you might have.

BTW, did you remove the circuit example you submitted?  For this circuit there was no load and so the force was the same.

hrtlacek, 2023-08-14:
yes, i completely changed my initial comment since I realized I just made a mistake that didn't have anything to do with mechanical simulation (same behavior with electrical components). Sorry for that.

I am super interested in the whole mechanical and acoustic analogy, but i am not an expert.
I am surprised that you say 'you have to draw with one topology and simulate with another'. Can you give literature/links on that topic? 
Do you mean  you are internally converting series to parallel etc before simulating?

mph-, 2023-08-15:
Hi I've pushed a patch to switch to the other analog (the mobility analog).  It seems to work but I have not fully tested it.  This analog is more intuitive except that mechanical impedance is equivalent to electrical impedance.  But it has the advantage that the circuit topology does not need to change.

There is a goodish comparison at https://en.wikipedia.org/wiki/Mechanical%E2%80%93electrical_analogies

mph-, 2024-02-12:
Oops, I meant to say that with the mobility analog, electrical impedance is not equivalent to mechanical impedance.

closed 2024-02-12 by mph-