mph-/lcapy #112: MNA equations example does not run

timchinowsky, 2023-10-30:
`lcapy` looks great, considering having students in the lab class I am currently teaching use it to gain insights into simulation.  Trying to derive and display the MNA equations as shown in https://lcapy.readthedocs.io/en/latest/circuits.html#networks-and-netlists, not having any luck, seems like the technique has changed.  Attaching a zipped Jupyter notebook showing my attempts.  New to `lcapy`, am I missing something simple?  Thanks!

[mna_equations.zip](https://github.com/mph-/lcapy/files/13210310/mna_equations.zip)

mph-, 2023-10-31:
Thanks for letting me know.  It seems like I inadvertently removed it when cleaning up some code!   Anyway, I have pushed a patch to restore it.

timchinowsky, 2023-10-31:
Wonderful, thanks!

closed 2023-10-31 by timchinowsky