mph-/lcapy #113: inspect.getmembers() crashes on equation object

timchinowsky, 2023-11-01:
Hi, not sure what this is about, but the standard python `inspect.getmembers()` crashes on an equation object returned by `matrix_equations()`:

```
from lcapy import Circuit
import inspect
c = Circuit("""
Vi 1 0 step; down
W 1 1_1; right
W 0 0_1; right
Rs 1_1 0_1; down
""")
eq = c.nodal_analysis().matrix_equations()
inspect.getmembers(eq)`
```

returns

```
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
Cell In[4], line 10
      3 c = Circuit("""
      4 Vi 1 0 step; down
      5 W 1 1_1; right
      6 W 0 0_1; right
      7 Rs 1_1 0_1; down
      8 """)
      9 eq = c.nodal_analysis().matrix_equations()
---> 10 inspect.getmembers(eq)

File ~/anaconda3/lib/python3.11/inspect.py:595, in getmembers(object, predicate)
    592 def getmembers(object, predicate=None):
    593     """Return all members of an object as (name, value) pairs sorted by name.
    594     Optionally, only return members that satisfy a given predicate."""
--> 595     return _getmembers(object, predicate, getattr)

File ~/anaconda3/lib/python3.11/inspect.py:573, in _getmembers(object, predicate, getter)
    568 for key in names:
    569     # First try to get the value via getattr.  Some descriptors don't
    570     # like calling their __get__ (see bug #1785), so fall back to
    571     # looking in the __dict__.
    572     try:
--> 573         value = getter(object, key)
    574         # handle the duplicate key
    575         if key in processed:

File ~/anaconda3/lib/python3.11/site-packages/lcapy/texpr.py:75, in TimeDomainExpression.abs(self)
     71 @property
     72 def abs(self):
     73     """Return absolute value."""
---> 75     return self.__class__(abs(self.expr), **self.assumptions)

TypeError: bad operand type for abs(): 'Equality'
```

mph-, 2023-11-02:
I've made a fix for this.  However, there is some collateral damage that I need to sort out before I push a patch.

timchinowsky, 2023-11-02:
Cool, thanks.  From the error messages, it seemed like the issue might be with `sympy` as well, but now I'm not sure.  I'm getting an error when inspecting a `sympy` matrix expression generated by `lcapy`, but when I generate what is seemingly the same matrix using only `sympy` it works fine.  There must be some differences under the hood I am not reproducing.

```
from lcapy import Circuit
import inspect
import sympy as sym
from sympy.core.symbol import Symbol

s = Symbol('V1')
m = sympy.Matrix([[Symbol('Vn1')], [Symbol('Vn2')], [Symbol('Vn3')], [Symbol('IV1')], [Symbol('IL1')]])

# this works
print("Sympy:", m)
inspect.getmembers(m)

a = Circuit("""
V1 1 0 {10 + v(t)}; down
R1 1 2; right
L1 2 3; right=1.5, i={i_L}
R2 3 0_3; down=1.5, i={i_{R2}}, v={v_{R2}}
W 0 0_3; right
W 3 3_a; right
C1 3_a 0_4; down, i={i_C}, v={v_C}
W 0_3 0_4; right""")

eq = a.dc().matrix_equations()
lhs = eq.expr.lhs

# this fails, on what seems like a similar expression
print("Lcapy:", lhs)
inspect.getmembers(lhs)

print("done")
```
output:
```
Sympy: Matrix([[Vn1], [Vn2], [Vn3], [IV1], [IL1]])
Lcapy: Matrix([[Vn1], [Vn2], [Vn3], [IV1], [IL1]])
---------------------------------------------------------------------------
NonSquareMatrixError                      Traceback (most recent call last)
Cell In[97], line 29
     27 # this fails, on what seems like a similar expression
     28 print("Lcapy:", lhs)
---> 29 inspect.getmembers(lhs)
     31 print("done")

File ~/anaconda3/lib/python3.11/inspect.py:595, in getmembers(object, predicate)
    592 def getmembers(object, predicate=None):
    593     """Return all members of an object as (name, value) pairs sorted by name.
    594     Optionally, only return members that satisfy a given predicate."""
--> 595     return _getmembers(object, predicate, getattr)

File ~/anaconda3/lib/python3.11/inspect.py:573, in _getmembers(object, predicate, getter)
    568 for key in names:
    569     # First try to get the value via getattr.  Some descriptors don't
    570     # like calling their __get__ (see bug #1785), so fall back to
    571     # looking in the __dict__.
    572     try:
--> 573         value = getter(object, key)
    574         # handle the duplicate key
    575         if key in processed:

File ~/anaconda3/lib/python3.11/site-packages/sympy/matrices/expressions/matexpr.py:274, in MatrixExpr.I(self)
    272 @property
    273 def I(self):
--> 274     return self.inverse()

File ~/anaconda3/lib/python3.11/site-packages/sympy/matrices/expressions/matexpr.py:262, in MatrixExpr.inverse(self)
    260 def inverse(self):
    261     if not self.is_square:
--> 262         raise NonSquareMatrixError('Inverse of non-square matrix')
    263     return self._eval_inverse()

NonSquareMatrixError: Inverse of non-square matrix
```

mph-, 2023-11-02:
The problem is not with SympPy; well it caused the problem by no longer allowing Lcapy objects to be stored in a SymPy matrix.  So, I'm having to rewrite the Lcapy matrix classes.

mph-, 2023-11-03:
I've pushed a patch but it is likely I have broken something else...

timchinowsky, 2023-11-03:
Yeah, the nice equation formatting in notebooks is one thing.  And, inspect still crashes, but in a different way.  Here's the code I'm looking at:

```
from lcapy import Circuit
import inspect

a = Circuit("""
V1 1 0 {10 + v(t)}; down
R1 1 2; right
L1 2 3; right=1.5, i={i_L}
R2 3 0_3; down=1.5, i={i_{R2}}, v={v_{R2}}
W 0 0_3; right
W 3 3_a; right
C1 3_a 0_4; down, i={i_C}, v={v_C}
W 0_3 0_4; right""")

eq = a.dc().matrix_equations()
display(eq)
inspect.getmembers(eq)
```

Maybe better to revert it, and just be less nosy with inspect for now...

mph-, 2023-11-03:
I think I have fixed it; it was really subtle.

timchinowsky, 2023-11-03:
I can imagine, there's a lot of stuff there, that I wish I understood better.  I am seeing that those commits fix the crash, but the equations still don't format TeX-like.  Expected?

mph-, 2023-11-03:
Okay, that must be for Jupyter.  I'll add a hook.

mph-, 2023-11-04:
Can you try this please?

timchinowsky, 2023-11-09:
It seems like inspect might cause a lot of errors in various places because there are a lot of attributes which throw errors if they are inappropriate for a given object.  I don't know if this really a bug or just a limitation of inspect, but thought I'd mention it.  The reason I started asking this question is because I was trying to systematically explore the `lcapy` objects and what they can do.  Here's one example:

```
from lcapy import Circuit

d = Circuit("""
C 1 2; right
L 2 0_2; down 
R 2 3; right
W 0 0_2; right
W 0_2 0_3; right
""")

d.draw()

ins = inspect.getmembers(d)
```

mph-, 2023-11-10:
Using inspect looks like a good way to check for bugs.  However, your network doesn't have a state-space realization and so Lcapy throws an exception.

closed 2023-12-30 by timchinowsky