mph-/lcapy #114: Incidence matrix?

timchinowsky, 2023-11-03:
Hey, not really an issue, but wondering if `lcapy` has a way of generating circuit incidence/reduced incidence matrices a la [Chua et al.](http://wla.berkeley.edu/~ee100/fa08/lectures/EE100supplementary_notes_2_pp21to33.pdf)?  I can see that the info is there to readily assemble the matrices, wondering if that is implemented somewhere.

mph-, 2023-11-04:
This is not currently done but should be straightforward with the machinery in CircuitGraph.

timchinowsky, 2023-11-04:
Wow great, thanks!  I'll play around with this.  Digging into networkx looks like that could also be used:  https://networkx.org/documentation/stable/reference/generated/networkx.linalg.graphmatrix.incidence_matrix.html

closed 2023-11-04 by timchinowsky

mph-, 2023-11-04:
I'm curious what you want to use an incidence matrix for.  I've seen them mentioned but have never had an application.