mph-/lcapy #115: Domain equation/solution inconsistencies

timchinowsky, 2023-11-04:
Looking at matrix_equations for various domains, I'm seeing some puzzling inconsistencies, wondering if I am understanding domains:

```
from lcapy import Circuit
from IPython.core.display import HTML

c = Circuit("""
V1 1 0 DC; down
R1 1 2; right
R2 2 0_1; down
W 0_1 0; left""")
c.draw()
c_eq = c.dc().matrix_equations()

# DC:  Everything seems consistent
display(HTML("<hr>DC Equations:"), c_eq)
display(HTML("<hr>DC simplified rhs:"), c_eq.rhs.expand().simplify())
display(HTML("<hr>DC v(R2):"), c.R2.v.simplify())

b = Circuit("""
V1 1 0 AC; down
R1 1 2; right
R2 2 0_1; down
W 0_1 0; left""")
b_eq = b.ac().matrix_equations()

# AC: v(R2) calculated ok, but RHS of equations is zero?
display(HTML("<hr>AC Equations:"), b_eq)
display(HTML("<hr>AC simplified rhs:"), b_eq.rhs.expand().simplify())
display(HTML("<hr>AC v(R2):"), b.R2.v.simplify())

a = Circuit("""
V1 1 0 {V(s)}; down
R1 1 2; right
R2 2 0_1; down
W 0_1 0; left""")
a_eq = a.laplace().matrix_equations()

# Laplace: RHS calculated correctly as function of V(s), but v(R2) shown as function of V(t)?
display(HTML("<hr>Laplace equations:"), a_eq)
display(HTML("<hr>Laplace simplified rhs:"), a_eq.rhs.expand().simplify())
display(HTML("<hr>Laplace v(R2):"), a.R2.v.simplify())
```

[image]

timchinowsky, 2024-01-03:
This issue seems to be related to my confusion with both `v` and `V` being properties of e.g. `R2`... I'll see if I can work up a consistent set of examples

mph-, 2024-02-12:
Did you manage to find a set of examples?