mph-/lcapy #119: Voltage sources crash transfer()

timchinowsky, 2023-11-29:
Not sure what expected behavior is for calculating transfer function of networks including voltage sources.  The help implies that they are ignored:

```
transfer(N1p, N1m, N2p=None, N2m=None) method of lcapy.circuit.Circuit instance
    Create Laplace-domain voltage transfer function V2(s) / V1(s) where:
    V1 is V[N1p] - V[N1m]
    V2 is V[N2p] - V[N2m]
    
    Note, independent sources are killed and initial conditions
    are ignored.
    
    Alternative forms are:
        transfer(N1p, N1m, N2p, N2m)
        transfer(cpt1, cpt2)
        transfer((N1p, N1m), cpt2)
        transfer(cpt1, (N2p, N2m))
```

But `transfer` fails for the following network which contains a voltage source, and succeeds if the voltage source is removed:

```
from lcapy import Circuit, Vector, s
import sympy
from lcapy import state; state.current_sign_convention = 'passive'

 # {V(s)}
d = Circuit("""
Ri 1 2; right 
C1 2 3; right
L 3 0_3; down
C2 3 4; right
RL 4 0_4; down
W 0_4 0_3; left
W 0_3 0_2; left
W 0_2 0; left
""")

d.draw()
d.describe()
display(d.transfer(1,0,4,0).simplify())

d = Circuit("""
Vi 1 0 {V(s)}; down
Ri 1 2; right 
C1 2 3; right
L 3 0_3; down
C2 3 4; right
RL 4 0_4; down
W 0_4 0_3; left
W 0_3 0_2; left
W 0_2 0; left
""")

d.draw()
d.describe()
help(d.transfer)
d.transfer(1,0,4,0).simplify()
```

The second call to `transfer()` crashes, is this expected?

timchinowsky, 2023-11-29:
And if it breaks because transfer() kills the voltage source by changing it to a short circuit, what would be the way to get the transfer function from the circuit with the voltage source still in it?  How would one calculate V(RL)/V(Vi)?

mph-, 2023-12-08:
Yes, `transfer`, replaces voltage sources with short-circuits so you need to remove the voltage source, for example,

    d.remove('V1').transfer(1, 0, 4, 0)

Alternatively,

    d.RL.V(s) / d.Vi.V(s)

mph-, 2023-12-08:
I suppose I could automatically remove any voltage sources across the input to the circuit, apply the test source to find the transfer function, and then restore the voltage sources...

timchinowsky, 2023-12-15:
Nice!  How does the PyPI upgrade process work?  I am seeing your changes appear pretty quickly, but not this one yet.  Are they gated by version number, or...?

timchinowsky, 2023-12-30:
I am finding that `pip install git+https://github.com/mph-/lcapy.git#egg=lcapy --upgrade --force` seems to work to get latest changes.  Needed to add the `--force`.  Closing...

closed 2023-12-30 by timchinowsky

mph-, 2023-12-30:
I'm not sure why you need `--force`.  Is there a dependency problem?

timchinowsky, 2024-01-02:
From `%pip install git+https://github.com/mph-/lcapy.git#egg=lcapy` I get

```
Collecting lcapy
  Cloning https://github.com/mph-/lcapy.git to /tmp/pip-install-y_i7xp7x/lcapy_dd6b865bee0a4b0ab9db1d8f30f46da9
  Running command git clone --filter=blob:none --quiet https://github.com/mph-/lcapy.git /tmp/pip-install-y_i7xp7x/lcapy_dd6b865bee0a4b0ab9db1d8f30f46da9
  Resolved https://github.com/mph-/lcapy.git to commit 32c8357a024087ef24b761c29061a336c5f58f8b
  Preparing metadata (setup.py) ... done
Requirement already satisfied: matplotlib in /home/timc/anaconda3/lib/python3.11/site-packages (from lcapy) (3.8.2)
Requirement already satisfied: scipy in /home/timc/anaconda3/lib/python3.11/site-packages (from lcapy) (1.11.4)
Requirement already satisfied: numpy in /home/timc/anaconda3/lib/python3.11/site-packages (from lcapy) (1.26.2)
Requirement already satisfied: sympy>=1.10.1 in /home/timc/anaconda3/lib/python3.11/site-packages (from lcapy) (1.12)
Requirement already satisfied: networkx in /home/timc/anaconda3/lib/python3.11/site-packages (from lcapy) (3.2.1)
Requirement already satisfied: IPython in /home/timc/anaconda3/lib/python3.11/site-packages (from lcapy) (8.19.0)
Requirement already satisfied: setuptools in /home/timc/anaconda3/lib/python3.11/site-packages (from lcapy) (69.0.3)
Requirement already satisfied: wheel in /home/timc/anaconda3/lib/python3.11/site-packages (from lcapy) (0.42.0)
Requirement already satisfied: property_cached in /home/timc/anaconda3/lib/python3.11/site-packages (from lcapy) (1.6.4)
Requirement already satisfied: mpmath>=0.19 in /home/timc/anaconda3/lib/python3.11/site-packages (from sympy>=1.10.1->lcapy) (1.3.0)
Requirement already satisfied: decorator in /home/timc/anaconda3/lib/python3.11/site-packages (from IPython->lcapy) (5.1.1)
Requirement already satisfied: jedi>=0.16 in /home/timc/anaconda3/lib/python3.11/site-packages (from IPython->lcapy) (0.19.1)
Requirement already satisfied: matplotlib-inline in /home/timc/anaconda3/lib/python3.11/site-packages (from IPython->lcapy) (0.1.6)
Requirement already satisfied: prompt-toolkit<3.1.0,>=3.0.41 in /home/timc/anaconda3/lib/python3.11/site-packages (from IPython->lcapy) (3.0.43)
Requirement already satisfied: pygments>=2.4.0 in /home/timc/anaconda3/lib/python3.11/site-packages (from IPython->lcapy) (2.17.2)
Requirement already satisfied: stack-data in /home/timc/anaconda3/lib/python3.11/site-packages (from IPython->lcapy) (0.6.3)
Requirement already satisfied: traitlets>=5 in /home/timc/anaconda3/lib/python3.11/site-packages (from IPython->lcapy) (5.14.0)
Requirement already satisfied: pexpect>4.3 in /home/timc/anaconda3/lib/python3.11/site-packages (from IPython->lcapy) (4.9.0)
Requirement already satisfied: contourpy>=1.0.1 in /home/timc/anaconda3/lib/python3.11/site-packages (from matplotlib->lcapy) (1.2.0)
Requirement already satisfied: cycler>=0.10 in /home/timc/anaconda3/lib/python3.11/site-packages (from matplotlib->lcapy) (0.12.1)
Requirement already satisfied: fonttools>=4.22.0 in /home/timc/anaconda3/lib/python3.11/site-packages (from matplotlib->lcapy) (4.47.0)
Requirement already satisfied: kiwisolver>=1.3.1 in /home/timc/anaconda3/lib/python3.11/site-packages (from matplotlib->lcapy) (1.4.5)
Requirement already satisfied: packaging>=20.0 in /home/timc/anaconda3/lib/python3.11/site-packages (from matplotlib->lcapy) (23.2)
Requirement already satisfied: pillow>=8 in /home/timc/anaconda3/lib/python3.11/site-packages (from matplotlib->lcapy) (10.1.0)
Requirement already satisfied: pyparsing>=2.3.1 in /home/timc/anaconda3/lib/python3.11/site-packages (from matplotlib->lcapy) (3.1.1)
Requirement already satisfied: python-dateutil>=2.7 in /home/timc/anaconda3/lib/python3.11/site-packages (from matplotlib->lcapy) (2.8.2)
Requirement already satisfied: parso<0.9.0,>=0.8.3 in /home/timc/anaconda3/lib/python3.11/site-packages (from jedi>=0.16->IPython->lcapy) (0.8.3)
Requirement already satisfied: ptyprocess>=0.5 in /home/timc/anaconda3/lib/python3.11/site-packages (from pexpect>4.3->IPython->lcapy) (0.7.0)
Requirement already satisfied: wcwidth in /home/timc/anaconda3/lib/python3.11/site-packages (from prompt-toolkit<3.1.0,>=3.0.41->IPython->lcapy) (0.2.12)
Requirement already satisfied: six>=1.5 in /home/timc/anaconda3/lib/python3.11/site-packages (from python-dateutil>=2.7->matplotlib->lcapy) (1.16.0)
Requirement already satisfied: executing>=1.2.0 in /home/timc/anaconda3/lib/python3.11/site-packages (from stack-data->IPython->lcapy) (2.0.1)
Requirement already satisfied: asttokens>=2.1.0 in /home/timc/anaconda3/lib/python3.11/site-packages (from stack-data->IPython->lcapy) (2.4.1)
Requirement already satisfied: pure-eval in /home/timc/anaconda3/lib/python3.11/site-packages (from stack-data->IPython->lcapy) (0.2.2)
Note: you may need to restart the kernel to use updated packages.
```

Whereas from `%pip install git+https://github.com/mph-/lcapy.git#egg=lcapy --force` I get

```
Collecting lcapy
  Cloning https://github.com/mph-/lcapy.git to /tmp/pip-install-jtxqq_mb/lcapy_f3f5123bf0d84695bbcf438156b25f25
  Running command git clone --filter=blob:none --quiet https://github.com/mph-/lcapy.git /tmp/pip-install-jtxqq_mb/lcapy_f3f5123bf0d84695bbcf438156b25f25
  Resolved https://github.com/mph-/lcapy.git to commit 32c8357a024087ef24b761c29061a336c5f58f8b
  Preparing metadata (setup.py) ... done
Collecting matplotlib (from lcapy)
[... cut ...] .0 pure-eval-0.2.2 pygments-2.17.2 pyparsing-3.1.1 python-dateutil-2.8.2 scipy-1.11.4 setuptools-69.0.3 six-1.16.0 stack-data-0.6.3 sympy-1.12 traitlets-5.14.1 wcwidth-0.2.12 wheel-0.42.0
Note: you may need to restart the kernel to use updated packages.
```
which throws some errors which look to be related to some Anaconda stuff I'm not using.  
The first version doesn't seem to actually update `lcapy`, as is doesn't build the wheel.  But my understanding of Python package installation is fuzzy.

mph-, 2024-01-03:
I think the problem is due to pip changes.  Can you try the `-e` option?

timchinowsky, 2024-01-03:
using `-e` I can roll back to 1.19, like

`%pip install -e git+https://github.com/mph-/lcapy.git@V1.19#egg=lcapy `

which seems to work:
```
Obtaining lcapy from git+https://github.com/mph-/lcapy.git@V1.19#egg=lcapy
  Updating ./src/lcapy clone (to revision V1.19)
  Running command git fetch -q --tags
  Running command git reset --hard -q f1032e38d8d4bec84ca8fde9c48945917f6963a6
  Preparing metadata (setup.py) ... done
Requirement already satisfied: matplotlib in /home/timc/anaconda3/lib/python3.11/site-packages (from lcapy) (3.8.2)
Requirement already satisfied: scipy in /home/timc/anaconda3/lib/python3.11/site-packages (from lcapy) (1.11.4)
Requirement already satisfied: numpy in /home/timc/anaconda3/lib/python3.11/site-packages (from lcapy) (1.26.2)
Requirement already satisfied: sympy>=1.10.1 in /home/timc/anaconda3/lib/python3.11/site-packages (from lcapy) (1.12)
Requirement already satisfied: networkx in /home/timc/anaconda3/lib/python3.11/site-packages (from lcapy) (3.2.1)
Requirement already satisfied: IPython in /home/timc/anaconda3/lib/python3.11/site-packages (from lcapy) (8.19.0)
Requirement already satisfied: setuptools in /home/timc/anaconda3/lib/python3.11/site-packages (from lcapy) (69.0.3)
Requirement already satisfied: wheel in /home/timc/anaconda3/lib/python3.11/site-packages (from lcapy) (0.42.0)
Requirement already satisfied: property_cached in /home/timc/anaconda3/lib/python3.11/site-packages (from lcapy) (1.6.4)
Requirement already satisfied: mpmath>=0.19 in /home/timc/anaconda3/lib/python3.11/site-packages (from sympy>=1.10.1->lcapy) (1.3.0)
Requirement already satisfied: decorator in /home/timc/anaconda3/lib/python3.11/site-packages (from IPython->lcapy) (5.1.1)
Requirement already satisfied: jedi>=0.16 in /home/timc/anaconda3/lib/python3.11/site-packages (from IPython->lcapy) (0.19.1)
Requirement already satisfied: matplotlib-inline in /home/timc/anaconda3/lib/python3.11/site-packages (from IPython->lcapy) (0.1.6)
Requirement already satisfied: prompt-toolkit<3.1.0,>=3.0.41 in /home/timc/anaconda3/lib/python3.11/site-packages (from IPython->lcapy) (3.0.43)
Requirement already satisfied: pygments>=2.4.0 in /home/timc/anaconda3/lib/python3.11/site-packages (from IPython->lcapy) (2.17.2)
Requirement already satisfied: stack-data in /home/timc/anaconda3/lib/python3.11/site-packages (from IPython->lcapy) (0.6.3)
Requirement already satisfied: traitlets>=5 in /home/timc/anaconda3/lib/python3.11/site-packages (from IPython->lcapy) (5.14.1)
Requirement already satisfied: pexpect>4.3 in /home/timc/anaconda3/lib/python3.11/site-packages (from IPython->lcapy) (4.9.0)
Requirement already satisfied: contourpy>=1.0.1 in /home/timc/anaconda3/lib/python3.11/site-packages (from matplotlib->lcapy) (1.2.0)
Requirement already satisfied: cycler>=0.10 in /home/timc/anaconda3/lib/python3.11/site-packages (from matplotlib->lcapy) (0.12.1)
Requirement already satisfied: fonttools>=4.22.0 in /home/timc/anaconda3/lib/python3.11/site-packages (from matplotlib->lcapy) (4.47.0)
Requirement already satisfied: kiwisolver>=1.3.1 in /home/timc/anaconda3/lib/python3.11/site-packages (from matplotlib->lcapy) (1.4.5)
Requirement already satisfied: packaging>=20.0 in /home/timc/anaconda3/lib/python3.11/site-packages (from matplotlib->lcapy) (23.2)
Requirement already satisfied: pillow>=8 in /home/timc/anaconda3/lib/python3.11/site-packages (from matplotlib->lcapy) (10.2.0)
Requirement already satisfied: pyparsing>=2.3.1 in /home/timc/anaconda3/lib/python3.11/site-packages (from matplotlib->lcapy) (3.1.1)
Requirement already satisfied: python-dateutil>=2.7 in /home/timc/anaconda3/lib/python3.11/site-packages (from matplotlib->lcapy) (2.8.2)
Requirement already satisfied: parso<0.9.0,>=0.8.3 in /home/timc/anaconda3/lib/python3.11/site-packages (from jedi>=0.16->IPython->lcapy) (0.8.3)
Requirement already satisfied: ptyprocess>=0.5 in /home/timc/anaconda3/lib/python3.11/site-packages (from pexpect>4.3->IPython->lcapy) (0.7.0)
Requirement already satisfied: wcwidth in /home/timc/anaconda3/lib/python3.11/site-packages (from prompt-toolkit<3.1.0,>=3.0.41->IPython->lcapy) (0.2.12)
Requirement already satisfied: six>=1.5 in /home/timc/anaconda3/lib/python3.11/site-packages (from python-dateutil>=2.7->matplotlib->lcapy) (1.16.0)
Requirement already satisfied: executing>=1.2.0 in /home/timc/anaconda3/lib/python3.11/site-packages (from stack-data->IPython->lcapy) (2.0.1)
Requirement already satisfied: asttokens>=2.1.0 in /home/timc/anaconda3/lib/python3.11/site-packages (from stack-data->IPython->lcapy) (2.4.1)
Requirement already satisfied: pure-eval in /home/timc/anaconda3/lib/python3.11/site-packages (from stack-data->IPython->lcapy) (0.2.2)
Installing collected packages: lcapy
  Attempting uninstall: lcapy
    Found existing installation: lcapy 1.21.dev0
    Uninstalling lcapy-1.21.dev0:
      Successfully uninstalled lcapy-1.21.dev0
  Running setup.py develop for lcapy
Successfully installed lcapy-1.19
Note: you may need to restart the kernel to use updated packages.
```
and then even without `-e` I can update to latest:

`%pip install git+https://github.com/mph-/lcapy.git#egg=lcapy`

```
Collecting lcapy
  Cloning https://github.com/mph-/lcapy.git to /tmp/pip-install-8467483a/lcapy_7accfe87073040ccb9f21b66eb0f3192
  Running command git clone --filter=blob:none --quiet https://github.com/mph-/lcapy.git /tmp/pip-install-8467483a/lcapy_7accfe87073040ccb9f21b66eb0f3192
[... cut ...]
  Stored in directory: /tmp/pip-ephem-wheel-cache-wyexwhws/wheels/d3/35/b3/7c17eb18690df7acc22c12c822e6c94251918a1b19e04d0a0d
Successfully built lcapy
Installing collected packages: lcapy
  Attempting uninstall: lcapy
    Found existing installation: lcapy 1.19
    Uninstalling lcapy-1.19:
      Successfully uninstalled lcapy-1.19
Successfully installed lcapy-1.21.dev0
Note: you may need to restart the kernel to use updated packages.
```
which all seems fine.  Next time you push a commit I'll try again and see what happens.  What do you expect from `-e`?