mph-/lcapy #120: Compact over-and-under display of component reference designator and value

timchinowsky, 2023-12-21:
By default `lcapy` displays components with both a reference designator and a value in `refdes=value` format like

[image]

Is it possible to configure the rendering so that the value is shown underneath the reference designator, without the equal sign, for a more compact display, for example
[image]

mph-, 2023-12-22:
You can achieve this using the 'l' and 'a' attributes for each component but that is a nuisance.
It should be straightforward to add a label style.  The hard part is thinking up a name for each style.  Do you have any suggestions?

timchinowsky, 2023-12-26:
Hmm... how about "=" and "stacked"?  Or "h" and "v" if that's not too generic.

mph-, 2023-12-29:
Well that turned out to be a lot harder than I thought.  Anyway, I've pushed a patch to master to control the component label style.  This is achieved with the `label_delimiter` attribute.
The default is `label_delimiter==`.  Your preferred style can be achieved with `label_delimiter=\\'.  You can also have the component name and component value on opposite sides of a component with `label_delimiter=a`.  See the new docs https://lcapy.readthedocs.io/en/latest/schematics.html :-)

mph-, 2023-12-29:
After some thought, my proposed suggestion is flexible but not obvious.  I think I may add a `label_style` attribute with options `stacked`, `split`, and maybe `aligned` (default).  This could be extended with `name` to only show the component name, and `value` to only show the component value.

timchinowsky, 2023-12-29:
Amazing, thank you, looks great!

closed 2023-12-29 by timchinowsky