mph-/lcapy #122: matrix_equations() fails on network with dependent source

timchinowsky, 2023-12-30:
This code for a passive network works:
```
from lcapy import Circuit
a = Circuit("""
V1 1 0 ac; down
R1 1 2 1; right
L1 2 3 1; right=1.5
R2 3 0_3 1; down=1.5
C1 3_a 0_4 1; down
W 0 0_3; right
W 3 3_a; right
W 0_3 0_4; right
""")
a.draw()
eq = a.ac().matrix_equations()
display(eq)
```
Here is the circuit:
[image]

But this code, which has a controlled source, does not:
```
from lcapy import Circuit
cct = Circuit(
"""
V1 1 0_6 ac; down
R1 1 2 10000; right=2
C1 3 0 1.1e-9; down=2.5
R2 2 3 10000; down
C2 2 4_5 2.2e-9; right=2
E1 4_8 0_3 opamp 3_1 4_6 100000;
RL 4 0_3 1e3; down

W 0_1 0; right
W 0 0_3; right
W 3 3_1; right=0.5
W 4_5 4; right
W 0_6 0_1; right
W 4_7 4_1 ; right
W 4_8 4_1; down
W 4_6 4_7; down
W 4_5 4_8; down
;label_style=stacked
""")
cct.draw()
display(cct.transfer(1,0,4,0))
cct.ac().matrix_equations()
```
It calculates the transfer function correctly, but errors out on the call to `matrix_equations()`.  Here is the circuit:

[image]

mph-, 2023-12-31:
I've pushed a fix for this.  However, I haven't managed to track down why SymPy takes so long...

timchinowsky, 2024-01-03:
Thanks!  Seems to work, but yes, slow.

closed 2024-01-03 by timchinowsky