mph-/lcapy #130: how to analysis circuit(multiple switches with different activation times)

YaoLiMuMu, 2024-04-01:
[image]
```pythonfrom lcapy import *
cct = Circuit("""
V 2_1 0 12; up, l=${U_1=12V}
SW0 2_1 2_0 nc 100; down, l=S_0(option)
R1 2_0 3_0 1e3; down
R2 3_0 4_0 1e4; right, l=${R_1^{\prime}=10\mathrm{~k}\Omega}
W 3_0 5_0; down=0.7
W 4_0 6_0; down=0.7
SW1 5_0 6_0 no 3; right, l=S_1
W 4_0 8_0; right
W 8_0 8_1; down=0.4
W 8_1 8_2; left=0.2, output=1.5, l=Test 1, aspect=1.5
VD1 8_0 12_0 0; right, invisible
D1 8_0 12_0; nosim, right, l=D1(option)
W 12_0 12_1; down=0.4
W 12_1 12_2; right=0.2, output=1.5, l=Test 2, aspect=1.5
R3 12_0 0 1e5; up, l=${R_3^{\prime}=100\mathrm{~k}\Omega}
W 12_0 9_0; right
W 9_0 11_0; right
SW2 9_0 13_0 no 1; up, l=S2^{\prime}(option)
R5 13_0 0 2740; up, l=${R_4^{\prime}=2.74\mathrm{~k}\Omega}
SW3 11_0 14_0 no 2; up, l=S2
R4 14_0 0 1300; up
; autoground=ground, label_style=aligned, node_spacing=4, scale=0.5, style=european, bipole label style={color=blue}
""")
cct.draw('/home/madhouse/Documents/Wolfram Mathematica/temple/appendix_c.png')
swtime = cct.convert_IVP(0)
str((swtime.R3.v + swtime.VD1.v).evalf(4))+',' + str(swtime.R3.v.evalf(4))
```

mph-, 2024-04-02:
I presume you want a different label for each switch.  You can do this with

    SW1 1 2; l=t\ge 0
    SW2 2 3; l=t\ge 1

[sw.pdf](https://github.com/mph-/lcapy/files/14833805/sw.pdf)

YaoLiMuMu, 2024-04-02:
> I presume you want a different label for each switch. You can do this with
> 
> ```
> SW1 1 2; l=t\ge 0
> SW2 2 3; l=t\ge 1
> ```
> 
> [sw.pdf](https://github.com/mph-/lcapy/files/14833805/sw.pdf)

not, I want the voltage  between node 8_0 and ground with differentes switch active at different times
> If there are multiple switches with different activation times, the initial values are evaluated recursively
I find the above documentation, but I don't how to use convert_IVP() method recursively, Can you give me an example

mph-, 2024-04-02:
Search the Lcapy docs for "setting initial values".

closed 2024-04-03 by YaoLiMuMu

mph-, 2024-04-04:
Yes, the SymPy parser does not like indexing in an expression.

You could remove `{sw_list[1]}` from the netlist and then use

    cct.subs('SW1', sw_list[1])

YaoLiMuMu, 2024-04-04:
@mph- Thank you very much for your reply. I found that using subs() requires components not to set initial values, and the switch component cannot use subs() either. I have found a solution and hope to help others in need.
The basic method is to use custom placeholders to set component parameters, and then use replace() or format() for string replacement. Here is an example:
```python
schdoc = """
V 2_1 0 12; up, l=${U_1=12V}
SW0 2_1 2_0 nc oo; down, l=S_0(option)
R1 2_0 3_0 1e3; down
R2 3_0 4_0 1e4; right, l=${R_1^{\prime}=10\mathrm{~k}\Omega}
W 3_0 5_0; down=0.7
W 4_0 6_0; down=0.7
SW1 5_0 6_0 no ???; right, l=S_1
W 4_0 8_0; right
W 8_0 8_1; down=0.4
W 8_1 8_2; left=0.2, output=1.5, l=Test 1, aspect=1.5
VD1 8_0 12_0 0; right, invisible
D1 8_0 12_0; nosim, right, l=D1(option)
W 12_0 12_1; down=0.4
W 12_1 12_2; right=0.2, output=1.5, l=Test 2, aspect=1.5
R3 12_0 0 1e5; up, l=${R_3^{\prime}=100\mathrm{~k}\Omega}
W 12_0 9_0; right
W 9_0 11_0; right
SW2 9_0 13_0 no ???; up, l=S_2^{\prime}(option)
R5 13_0 0 2740; up, l=${R_4^{\prime}=2.74\mathrm{~k}\Omega}
SW3 11_0 14_0 no ???; up, l=S_2
R4 14_0 0 1300; up
; autoground=ground, label_style=aligned, node_spacing=4, scale=0.5, style=european, bipole label style={color=blue}
"""
schdoc.replace("{", "\left_brace").replace("}", "\right_brace").replace("???", "{}").format(list[0], list[1], list[2],).replace("\left_brace", "{").replace("\right_brace", "}")
cct = Circuit(schdoc)
```
In the example above, because "{}" is used for other purposes, I use "???" as a placeholder.