mph-/lcapy #132: Support for constant phase elements in circuit simulation

ma-sadeghi, 2024-06-11:
I was wondering how difficult it would be to add support for CPE elements for circuit simulations. Currently, the circuit simulator doesn't seem to support it.

mph-, 2024-06-12:
The stamps for modified nodal analysis should be straightforward to add.   I think I know how to handle the inverse Laplace stuff.  But, when all is said in done, I imagine that most results will be gnarly.  Are you comfortable with fractional derivatives?

ma-sadeghi, 2024-06-13:
I'm very new to fractional derivatives, but willing to dig deeper as I really need to get the V(t) response to an input I(t) for a circuit with a few CPE elements.

mph-, 2024-06-13:
I had forgotten that I had implement the MNA stamps for CPEs.  Thus you can get an answer in the Laplace and ac domains.  For example,

    >>> a = Circuit("""
    V 1 0 ac
    CPE 1 2 K alpha
    R 2 0""")
    >>> a.R.I(t)

                          ⎛       α      ⎞                        ⎛       α      ⎞
            α             ⎜      ⅉ       ⎟         α              ⎜      ⅉ       ⎟
    - K⋅V⋅ω₀ ⋅sin(ω₀⋅t)⋅iₘ⎜──────────────⎟ + K⋅V⋅ω₀  ⋅cos(ω₀⋅t)⋅rₑ⎜──────────────⎟
                          ⎜ α       α    ⎟                        ⎜ α       α    ⎟
                          ⎝ⅉ ⋅K⋅R⋅ω₀  + 1⎠                        ⎝ⅉ ⋅K⋅R⋅ω₀  + 1⎠

Unfortunately, SymPy crashes when trying to do an inverse Laplace transform of s^alpha so I'll need to implement this.

Are you modelling the impedance of electrodes?

ma-sadeghi, 2024-06-14:
Thanks for sharing the snippet.

Yes, I'm modeling the impedance of Li-ion battery electrodes. Currently, I want to take it one step further (and not just match simulated Z to experimental Z): Particularly, I want to see the mismatch between experimental V(t) -given a known I(t)- and simulated V(t).

mph-, 2024-06-14:
If you want the transient response we need the inverse Laplace transform of 1 / (s^a + b)
However, I only know the cases for integer a and a = 0.5.

mph-, 2024-06-21:
Lcapy can solve circuits numerically but you are better off using SPICE.  However, you will need to create an approximate model of a CPE.

Lcapy can create a second order Pade approximation of the impedance of a CPE.  For example,

    >>> a = CPE('K', 'alpha')
    >>> Z = a.Z.approximate('pade', 2)

ma-sadeghi, 2024-06-25:
I'm not familiar with SPICE (I found a Python wrapper called PySpice, is that what you're referring to?), and at this point I just need a proof of concept. I'll make sure to look into do it properly in SPICE if it turns out to work as I thought it would. In the meantime, Is there any workaround/hack to make it work in Lcapy? Thanks again :)

mph-, 2024-06-28:
Yes, you can use PySpice for a numerical solution.   As I said before, for a symbolic solution we need the inverse Laplace transform of 1 / (s^a + b), which I don't know, or approximate the s-domain result using a Pade approximation before doing an inverse Laplace transform.