mph-/lcapy #135: '.sympy' on an expressions causes Jupyter Notebook to hang

anaidan, 2024-07-01:
My notebook hands 100% of the time I try this:

from lcapy import *
a = 2 * t + 3
a.sympy

I tried not doing a * import, but no luck there either.  Any suggestions?

mph-, 2024-07-01:
Bother, this works for me.  What do you get running `show_version()` ?

anaidan, 2024-07-01:
Thanks for the quick response:

Python: 3.11.5 | packaged by Anaconda, Inc. | (main, Sep 11 2023, 13:26:23) [MSC v.1916 64 bit (AMD64)]
SymPy: 1.11.1
NumPy: 1.24.3
Matplotlib: 3.7.2
SciPy: 1.11.1
Lcapy: 1.23

mph-, 2024-07-01:
Would you mind trying to update SymPy?  I;m currently using 1.11.1

mph-, 2024-07-01:
Oops, that's your version!  I'm using 1.12

mph-, 2024-07-01:
To find where it hangs, you could try

    from lcapy import t
    t.pdb().sympy

and step through in the Python debugger...

hakanozlemis, 2024-11-20:
Hello, I am facing a similar issue. Do you have any updates or any workaround on this open issue?

@anaidan 
@mph-

mph-, 2024-11-20:
The problem is that I cannot replicate this on my computer.   Which OS are you using?

hakanozlemis, 2024-12-20:
I have a Windows PC and have the following package versions:
```
sympy: 1.13.0
lcapy: 1.24
jupyter: 1.0.0
notebook: 7.1.3
```

hakanozlemis, 2025-02-11:
@mph- @anaidan,  do you have any updates on this matter?