mph-/lcapy #137: circuit_graph.draw() TypeError

ninja-lab, 2024-08-28:
All the examples for drawing circuit_graphs are throwing an error. 
Using `lcapy 1.24` 
```
cct = Circuit("""
...V1 1 0; down
...R1 1 2; right
...L1 2 3; right
...R2 3 4; right
...L2 2 0_2; down
...C2 3 0_3; down
...R3 4 0_4; down
...W 0 0_2; right
...W 0_2 0_3; right
...W 0_3 0_4; right""")
cct.draw()
cg = cct.circuit_graph()
cg.draw()
```
```
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
Cell In[117], line 2
      1 cg = a.circuit_graph()
----> 2 cg.draw()

File [~/anaconda3/envs/py312/lib/python3.12/site-packages/lcapy/circuitgraph.py:260](http://localhost:8889/home/erik/anaconda3/envs/py312/lib/python3.12/site-packages/lcapy/circuitgraph.py#line=259), in CircuitGraph.draw(self, filename, axes)
    257 pos = nx.spring_layout(G)
    259 labels = dict(zip(G.nodes(), G.nodes()))
--> 260 nx.draw_networkx(G, pos, axes, labels=labels)
    262 edge_labels = dict([((u, v), d['name'])
    263                     for u, v, d in G.edges(data=True)])
    264 nx.draw_networkx_edge_labels(G, pos, edge_labels=edge_labels)

File [~/anaconda3/envs/py312/lib/python3.12/site-packages/networkx/drawing/nx_pylab.py:312](http://localhost:8889/home/erik/anaconda3/envs/py312/lib/python3.12/site-packages/networkx/drawing/nx_pylab.py#line=311), in draw_networkx(G, pos, arrows, with_labels, **kwds)
    309     pos = nx.drawing.spring_layout(G)  # default to spring layout
    311 draw_networkx_nodes(G, pos, **node_kwds)
--> 312 draw_networkx_edges(G, pos, arrows=arrows, **edge_kwds)
    313 if with_labels:
    314     draw_networkx_labels(G, pos, **label_kwds)

File [~/anaconda3/envs/py312/lib/python3.12/site-packages/networkx/drawing/nx_pylab.py:876](http://localhost:8889/home/erik/anaconda3/envs/py312/lib/python3.12/site-packages/networkx/drawing/nx_pylab.py#line=875), in draw_networkx_edges(G, pos, edgelist, width, edge_color, style, alpha, arrowstyle, arrowsize, edge_cmap, edge_vmin, edge_vmax, ax, arrows, label, node_size, nodelist, node_shape, connectionstyle, min_source_margin, min_target_margin, hide_ticks)
    874 else:
    875     if not isinstance(arrows, bool):
--> 876         raise TypeError("Argument `arrows` must be of type bool or None")
    877     use_linecollection = not arrows
    879 if isinstance(connectionstyle, str):

TypeError: Argument `arrows` must be of type bool or None
```

mph-, 2024-08-29:
Hmmm, it works for me.  It seems to be a problem with networkx.  Which version do you have installed?

ninja-lab, 2024-09-02:
I have `networkx 3.3`, with `python 3.12.4`. What about you?

mph-, 2024-09-02:
I have networkx 3.1  

I can't install 3.3 from pypi.   Can you try networkx 3.1 to confirm the source of the problem?

ninja-lab, 2024-09-15:
Confirmed, I just installed networkx 3.1 and now the circuit graphs all draw as expected.