mph-/lcapy #140: CCCS doesn't always works as expected

lcarpineto, 2024-09-21:
Hi, I've been trying to generate equations for circuits were I need to use CCCS. 
In the attached test bench I've tried to use as control source either V1 or R1. 
The current flowing V1 or R1 is always the same, but I do get different results. The V1 case is the correct one (apart from the fact that the sign is not what I would expect). In the case of R1 the answer is instead wrong.

```python
from lcapy import Circuit
c=Circuit("""
# Created by lcapy-tk V0.94
; nodes={1@(2.5, 8.5), 2@(2.5, 6.5), 3@(5.5, 8.5), 4@(8, 8.5), 5@(11, 8.5), 6@(11, 6.5), 0@(2.5, 5.5), 7@(13, 6.5), 8@(13, 8.5), 9@(15, 8.5), 10@(15, 6.5)}
W1 1 3; right=1.5
R1 3 4; right=1.25
W2 4 5; right=1.5
R2 5 6; down
W3 6 2; left=4.25
W4 2 0; down=0.5, scale=0.5, ground
W5 6 7; right
W6 8 9; right
W7 10 7; left
R3 9 10; down
V1 1 2; down
F1 8 7 V1 F1; down
; draw_nodes=connections, label_nodes=all, style=american, voltage_dir=RP, label_style=name """)
c.draw()
c.R3.I

```

[image]

replacing this line
F1 8 7 R1 F1; down

```python
from lcapy import Circuit
c=Circuit("""
# Created by lcapy-tk V0.94
; nodes={1@(2.5, 8.5), 2@(2.5, 6.5), 3@(5.5, 8.5), 4@(8, 8.5), 5@(11, 8.5), 6@(11, 6.5), 0@(2.5, 5.5), 7@(13, 6.5), 8@(13, 8.5), 9@(15, 8.5), 10@(15, 6.5)}
W1 1 3; right=1.5
R1 3 4; right=1.25
W2 4 5; right=1.5
R2 5 6; down
W3 6 2; left=4.25
W4 2 0; down=0.5, scale=0.5, ground
W5 6 7; right
W6 8 9; right
W7 10 7; left
R3 9 10; down
V1 1 2; down
F1 8 7 R1 F1; down
; draw_nodes=connections, label_nodes=all, style=american, voltage_dir=RP, label_style=name """)
c.draw()
c.R3.I
```
[image]

A further example where I see issue is when I try to reproduce the "CapMultiplier" scheme. The use of CCCS serve the purpouse of increasing the capacitance. The issue here is that the equations are not generated.
[image]

```python
cct=Circuit("""
# Created by lcapy-tk V0.94
; nodes={1@(2.5, 7.5), 2@(2.5, 5), 3@(2.5, 2.5), 4@(5, 7.5), 5@(5, 2.5), 0@(5, 1.5), 6@(7, 7.5), 7@(7, 2.5)}
C1 2 3; down=1.25
W1 1 4; right=1.25
W2 3 5; right=1.25
W3 5 0; down=0.5, scale=0.5, ground
F1 4 5 V1 F1; down=2.5
V1 1 2 ac V1 0; down=1.25
W4 4 6; right
I1 6 7 ac; down=2.5
W5 7 5; left
; draw_nodes=connections, label_nodes=all, style=american, voltage_dir=RP, label_style=name """)
cct.draw()

```

[image]

```python
cct.ac().matrix_equations()
```
NonInvertibleMatrixError: Matrix det == 0; not invertible.

Any help is very much appreciated.

``` 

``` 

```

mph-, 2024-09-22:
Thanks for letting me know about this.

I think the result using a voltage source for the controlling current is correct as per the SPICE definition of a CCCS where the control current is considered to flow from the positive node through the voltage source to the negative node.

I will have to think about the case where the control current flows through a resistor.  I agree Lcapy gives the wrong result.  The simplest option would be to only allow the control current to be defined for a voltage source, as per SPICE.

lcarpineto, 2024-09-22:
Thanks for looking into this.
I agree that the restriction to only Vsources as controlling devices is not a limitation.
Concerning the sign, I've double checked with LTSPICE and I do get a positive value for the current, not negative, so I still think there is a problem there. See the simulated results of a similar circuit and the associated netlist. (V1=1V, R1=R2=R3=1Ohm, F1=1 -->I=0.5A -->V(R3)=0.5V
[image]

Concerning the last topic, (i.e. obtaining the impedance of the capacitive multiplierI tried adding a parallel resistance to the cap then sending it lim-->oo).  I could get this trick to work on other tools like CircuitNAV or SapWin4, but I can't still get lcapy to produce the MNA equations. 

```python

cct=Circuit("""
# Created by lcapy-tk V0.94
; nodes={1@(2.5, 7.5), 2@(2.5, 5), 3@(2.5, 2.5), 4@(5, 7.5), 5@(5, 2.5), 0@(5, 1.5), 6@(7, 7.5), 7@(7, 2.5)}
C1 2 3; down=1.25
W1 1 4; right=1.25
W2 3 5; right=1.25
W3 5 0; down=0.5, scale=0.5, ground
F1 4 5 V1 F1; down=2.5
V1 1 2 0; down=1.25
W4 4 6; right
#I1 6 7 ac; down=2.5
W5 7 5; left
W 6 6_0; right
R00 6_0 0;down, ground
; draw_nodes=connections, label_nodes=all, style=american, voltage_dir=RP, label_style=name """)

```

mph-, 2024-09-22:
I have pushed a trivial fix for this.  However, it has raised a much greater problem where the positive and negative nodes of current sources are swapped compared to SPICE!  The source I used had the arrow the wrong way and I didn't check carefully enough.

I'm not sure of the best strategy to correct this anomaly.