mph-/lcapy #142: Gaps with potentiometers and non-default cpt_size or node_spacing

why-trv, 2024-10-03:
First of all, thank you so much for lcapy, it is awesome!

It seems that drawing breaks a little with potentiometers if non-default cpt_size or node_spacing is used.
For example, this circuit:
```VIN IN 0 AC; right, l=V_{IN}
RIN IN 2; right, l=R_{IN}
W 2 2_1; up=0.5
W 2 2_2; down=0.5
C1 2_1 4; right
R1 2_2 3; right
RVT 4 3 5; down, l_=R_T
R2 4 0; up
C2 3 0; down
W 5 OUT; right=0.5
RL OUT 0; down

Renders fine by default:
<img width="323" alt="Screenshot 2024-10-03 at 23 50 22" src="https://github.com/user-attachments/assets/a0f7b178-61dd-4b09-a807-553200ec704d">

But exhibits a gap with cpt_size=1.25:
<img width="320" alt="Screenshot 2024-10-03 at 23 51 29" src="https://github.com/user-attachments/assets/2f5f6271-4d56-41da-962f-f0d676134499">

Or node_spacing=3:
<img width="454" alt="Screenshot 2024-10-03 at 23 51 58" src="https://github.com/user-attachments/assets/7c2aba20-6808-48da-86c2-fde93d9d8b60">

mph-, 2024-10-04:
Hopefully this patch fixes it.

why-trv, 2024-10-04:
It does for me, thank you!

closed 2024-10-04 by why-trv

mph-, 2024-10-06:
I had a crack at supporting RV.  Now you can do something like...

    from lcapy import Circuit, t

    a = Circuit('''
    V 1 0
    RV 1 0 2 R a
    ''')

print(a[2].V(t))

why-trv, 2024-10-06:
That's great!

Meanwhile, something similar happens to opamps.

Default scaling:
[image]

Tweaked `node_spacing`:
[image]

mph-, 2024-10-07:
Ah, that's deliberate to honour the node spacing.  You can remove the steps by scaling the opamp, e.g., 

    E1 1 0 opamp 2 3; right, scale=1.5

why-trv, 2024-10-09:
OK, got it, thanks!

why-trv, 2024-10-09:
By the way (off-topic, but not sure this warrants an issue), is possible to define an id somewhere in the netlist that will come through as the svg path/group id for the component?

mph-, 2024-10-09:
I'm not sure if this is possible.  svg images are generated using `pdflatex` and then converted to svg using `pdf2svg`.