mph-/lcapy #145: Switching between symbolic and numeric analysis

why-trv, 2024-10-04:
If I create a circuit from a netlist with component values and, say, get its transfer function, lcapy is always going to use substitute in component values, or am I missing something and there's a way to switch to symbolic analysis?
If so, is there a built-in way to remove component values without having to replicate the component parsing logic that's already there (e.g. determining which is the component value based on what type of component it is)?

mph-, 2024-10-04:
I've been wanting this myself.  I've added  `sympify` method to remove the numerical values.  Would you mind testing it?  I still need to add support for opamps.

It might be useful to have a method that creates a mapping between symbolic and numeric values to recover the original netlist using the `subs` method.

why-trv, 2024-10-05:
Thank you so much! Works for me so far, with simple circuits of R and C.

why-trv, 2024-10-18:
`sympify()` seems to have a problem with embedded commands. For example, here's a netlist:
```
;; \ctikzset{inductor=american}
V vt vb AC; down
R1 vt r1r; right
L1 r1r l1r; right
L2 l1r l2r; right
L3 l2r l3r; right
W l3r c1t; right=0.5
C1 c1t c1b; down
W l1r l1r_; down=0.5
W l2r l2r_; down=0.5
W l3r l3r_; down=0.5
R2 l1r_ l2r_; right
R3 l2r_ l3r_; right
W vb c1b; right
W c1t OUT_V; right=0.25
W c1b OUT_G; right=0.25
```

```
Traceback (most recent call last):
  File "/whatever/resources/lcapy/process-circuit.py", line 658, in <module>
    main()
  File "/whatever/resources/lcapy/process-circuit.py", line 597, in main
    cct = circuit_for_drawing(cct)
          ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/whatever/resources/lcapy/process-circuit.py", line 455, in circuit_for_drawing
    return cct.sympify()    
           ^^^^^^^^^^^^^
  File "/whatever/.venv/lib/python3.12/site-packages/lcapy/netlistmixin.py", line 1083, in sympify
    net = cpt.sympify()
          ^^^^^^^^^^^^^
  File "/whatever/.venv/lib/python3.12/site-packages/lcapy/mnacpts.py", line 925, in sympify
    return self._netmake(args=[])
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/whatever/.venv/lib/python3.12/site-packages/lcapy/mnacpts.py", line 421, in _netmake
    return self._netmake1(self.namespace + self.relname, nodes, args, opts,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/whatever/.venv/lib/python3.12/site-packages/lcapy/mnacpts.py", line 401, in _netmake1
    if self.keyword[0] == 0 and self.keyword[1] != '':
       ~~~~~~~~~~~~^^^
IndexError: string index out of range
```

why-trv, 2024-10-18:
BTW, maybe there's a way to adjust global style per component type? All I'm trying to do really is to use 'american' style for inductors, while keeping resistors 'european'

mph-, 2024-10-18:
I've never considered a mixed style.   If you suggest a syntax, I will see if I can implement it.

mph-, 2024-10-19:
In the interim you can use:

    L1 1 2; right, kind=american

mph-, 2024-10-22:
You can also now do

    ; inductor_style=american, resistor_style=european

why-trv, 2024-10-24:
Awesome, thank you so much!