mph-/lcapy #147: ValueError: The MNA A matrix is not invertible for time analysis: The network is disjoint.  These nodes have no path to node 0: 3

Digiyang, 2024-10-08:
### Description
When attempting to convert a circuit to a two-port model using the `twoport()` method, I'm encountering a `NonInvertibleMatrixError`. The error suggests that the network is disjoint, but the circuit appears to be properly connected.

### Steps to Reproduce
1. Create the following circuit:
```python
import lcapy as lca

a = lca.Circuit("""
E 1 0 opamp 3 2 Ad; right
W 2 2_1; down
R1 2_1 7 R1; down
R2 2_1 1_1 R2; right
W 1 1_1; down
P1 4 5; down
W 4 3; right
W 1 6; right
P2 6 0; down
W 5 7; right
W 7 0; right
; draw_nodes=connections, label_ids=none, label_nodes=True
""")
```

2. Attempt to convert the circuit to a two-port model:
```python
a.twoport('P1', 'P2')
```

### Error Message
```
DMNonInvertibleMatrixError: Non-unique solution.

NonInvertibleMatrixError: Matrix det == 0; not invertible.

ValueError: The MNA A matrix is not invertible for time analysis: The network is disjoint. These nodes have no path to node 0: 3
```

### Additional Context
I've verified that all nodes in the circuit are connected, including node 3. The error persists even after ensuring all connections are present.

### Possible Bug
There might be an issue with how lcapy is handling the opamp in the circuit or how it's analyzing the network topology for the two-port conversion.

mph-, 2024-10-09:
It looks like there are three problems here.  

Firstly, there is no DC path to ground for node 4.  This can be achieved by adding a resistor to model the opamp's input resistance.  Later on you can evaluate the expression as this goes to infinity in the limit.

Secondly, the opamp has no output resistance and this will cause problems trying to determine two-port parameters by applying a voltage source across the output of the opamp.  You can add output resistance using:

    E 1 0 opamp 3 2 Ad 0 Ro; right

Thirdly, some two port models are not invertible.   By default, the `twoport()` method creates a B (inverse ABCD) model and I don't think this exists for your circuit.  You could try using`model=A` or `model=Z`.

Digiyang, 2024-10-11:
This works for me! Thank you for your help

closed 2024-10-17 by Digiyang