mph-/lcapy #153: Transfer function and gyrator

simonArchipoff, 2024-12-15:
Hello, not sure I understand what I am doing, I am trying to compute a transfer function of a circuit using a gyrator, but I can make it. Is it normal ?
I use lcapy 1.24.
My apologies if this post is non-sense, if you can point me to a reference then that would allow me to understand, that would be great)

```python

a = Circuit("""
    P1 1 0 ; down, v_=v_1
    GY1 4 5 0 1 {10}; right
    R1 4 5 {1} ; down
    W 5 0_1 ; down
    W 0 0_1; right
    """)

H = a.transfer('P1', 'R1')
```

Raise an error : 
```
  File "[…]python3.12/site-packages/lcapy/mna.py", line 344, in Vdict
    self._solve()
  File "[…]/python3.12/site-packages/lcapy/mna.py", line 259, in _solve
    results[m + num_nodes], cct.elements[key].is_source)
                            ~~~~~~~~~~~~^^^^^
KeyError: 'GY1X'
```

mph-, 2024-12-18:
Thanks for letting me know about this.  I  pushed a patch several days ago but tagged it with the incorrect issue number!

closed 2024-12-19 by simonArchipoff

mph-, 2024-12-19:
No problem.  I added some tests and it gives the correct results so I assume I have implemented it correctly.