mph-/lcapy #154: transformer and laplace mesh analysis

simonArchipoff, 2024-12-20:
Hello, with the main branch of lcapy I came across this issue :
(I might be doing some weird things because I use lcapy to do multi-physics simulation while learning how to do it)

```python
from lcapy import *
cct = Circuit("""
    Vs 0 1;
    TF1 0 1 2 3 {1};
    R1 2 3 1;
""")
foo = cct.laplace().mesh_analysis()
```

It raises an exception :

```
Traceback (most recent call last):
  File "main.py", line 11, in <module>
    foo = cct.laplace().mesh_analysis()
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/simon/.cache/pypoetry/virtualenvs/spykersim-6fZ2i_oL-py3.12/lib/python3.12/site-packages/lcapy/netlist.py", line 859, in mesh_analysis
    return LoopAnalysis.from_circuit(self)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/simon/.cache/pypoetry/virtualenvs/spykersim-6fZ2i_oL-py3.12/lib/python3.12/site-packages/lcapy/loopanalysis.py", line 57, in from_circuit
    return cls(cct)
           ^^^^^^^^
  File "/home/simon/.cache/pypoetry/virtualenvs/spykersim-6fZ2i_oL-py3.12/lib/python3.12/site-packages/lcapy/loopanalysis.py", line 74, in __init__
    self._equations = self._make_equations()
                      ^^^^^^^^^^^^^^^^^^^^^^
  File "/home/simon/.cache/pypoetry/virtualenvs/spykersim-6fZ2i_oL-py3.12/lib/python3.12/site-packages/lcapy/loopanalysis.py", line 190, in _make_equations
    result = self._process_loop(loop, mesh_currents[m],
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/simon/.cache/pypoetry/virtualenvs/spykersim-6fZ2i_oL-py3.12/lib/python3.12/site-packages/lcapy/loopanalysis.py", line 164, in _process_loop
    v = -elt.cpt.voltage_equation(current, self.kind)
         ^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'TF' object has no attribute 'voltage_equation'. Did you mean: 'voltage_gain'?
```

mph-, 2024-12-24:
Yes, this is a bug.  I will need to ponder a fix...

mph-, 2024-12-25:
There is no simple fix.  Indeed, I have never seen an example of mesh analysis with a transformer in a circuit.  I think it is possible but not easy to implement.  The mesh equations will need additional auxiliary equations relating the secondary to primary current for each transformer.

Do you have a specific requirement for this?  An alternative would be to get Lcapy to generate a system of equations using modified nodal analysis.

simonArchipoff, 2024-12-26:
Hello,
No, I dont have an actual example because I have no idea what I am doing…
I am monkey fuzz boxing lcapy :-)

mph-, 2024-12-26:
Fuzz away!  There are many untested hidden corners that I have not wandered down.